pvizeli/CmdParser #5: BUG parses empty buffers

tablatronix, 2016-12-16:
test
```
    CmdBuffer<32> myBuffer;
	if (cmdParser.parseCmd(&myBuffer) != CMDPARSER_ERROR) Serial.println("myBuffer not empty");
```

parses empty buffers without error, not sure if this is correct, cause then subsequent calls are useless
cmdParser.getCommand(); is empty
cmdParser.getCmdParam(1) throws exceptions even , if you are not checking for 0 param count first ( this might be a second bug that can be addressed to avoid exceptions.

tablatronix, 2016-12-17:
I understand why a NULL check is failing, it will never be null

But i would think

buffer == 0 would work but it does not, I guess we can just assume buffer[0] == 0x00 is empty and useless, but then you cannot parse empty newlines

tablatronix, 2016-12-17:
well I guess you cannot do that at all.

I can only think of doing memcmp with an empty buffer or a loop check or change the functionality of [0] == 0x00 to return something other than paramcnt

pvizeli, 2017-06-16:
Fix with https://github.com/pvizeli/CmdParser/commit/943873564c3ecbde18e52505bf12f3d7961192cd

closed 2017-06-16 by pvizeli